USC-ACTLab/crazyswarm #608: FAILED test_videoOutput.py::test_videoOutput --- error in visVispy.py

realm-robot-ws, 2022-03-07:
##  Summary

The test `test_videoOutput.py` is failing. The root cause appears to be an error in `pycrazyswarm/visualizer/visVispy.py` that prevents a valid mp4 file from being written. `test_videoOutput.py` then fails when trying to read this file, since it contains no streams.

## Steps to reproduce

1. Install following https://crazyswarm.readthedocs.io/en/latest/installation.html. Running `build.sh` is successful
2. Run unit tests:
```
cd ros_ws/src/crazyswarm/scripts
source ../../../devel/setup.bash
$CSW_PYTHON -m pytest
```

## Expected behavior

Tests pass successfully, and a window opens up during the test showing a crazyflie moving around.

## Actual behavior

`test_videoOutput.py` fails:
```
=============================================================================================== test session starts ===============================================================================================
platform linux -- Python 3.8.10, pytest-6.2.4, py-1.10.0, pluggy-0.13.1
rootdir: /home/realm/crazyflie/crazyswarm/ros_ws/src/crazyswarm/scripts, configfile: pytest.ini
plugins: anyio-3.5.0
collected 31 items                                                                                                                                                                                                

test_collisionAvoidance.py ............                                                                                                                                                                     [ 38%]
test_highLevel.py .........                                                                                                                                                                                 [ 67%]
test_lowLevel.py ......                                                                                                                                                                                     [ 87%]
test_ros.py .                                                                                                                                                                                               [ 90%]
test_simOnly.py .                                                                                                                                                                                           [ 93%]
test_videoOutput.py F                                                                                                                                                                                       [ 96%]
test_yamlString.py .                                                                                                                                                                                        [100%]

==================================================================================================== FAILURES =====================================================================================================
________________________________________________________________________________________________ test_videoOutput _________________________________________________________________________________________________

tmp_path = PosixPath('/tmp/pytest-of-realm/pytest-16/test_videoOutput0')

    @pytest.mark.skipif("TRAVIS" in os.environ or "CI" in os.environ,
                        reason="CI usually cannot create OpenGL context.")
    def test_videoOutput(tmp_path):
        # tmp_path is supplied by pytest - a directory where we can write that will
        # eventually be deleted.
        path = str(tmp_path / "crazyswarm_test_video.mp4")
        subprocess.call([os.environ["CSW_PYTHON"], __file__, path])
    
        import ffmpeg
        properties = ffmpeg.probe(path)
>       stream = properties["streams"][0]
E       IndexError: list index out of range

test_videoOutput.py:68: IndexError
---------------------------------------------------------------------------------------------- Captured stdout call -----------------------------------------------------------------------------------------------
Warning: No joystick found!
wrote 0 frames to /tmp/pytest-of-realm/pytest-16/test_videoOutput0/crazyswarm_test_video.mp4
---------------------------------------------------------------------------------------------- Captured stderr call -----------------------------------------------------------------------------------------------
ffmpeg version 4.2.4-1ubuntu0.1 Copyright (c) 2000-2020 the FFmpeg developers
  built with gcc 9 (Ubuntu 9.3.0-10ubuntu2)
  configuration: --prefix=/usr --extra-version=1ubuntu0.1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-nvenc --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 31.100 / 56. 31.100
  libavcodec     58. 54.100 / 58. 54.100
  libavformat    58. 29.100 / 58. 29.100
  libavdevice    58.  8.100 / 58.  8.100
  libavfilter     7. 57.100 /  7. 57.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  5.100 /  5.  5.100
  libswresample   3.  5.100 /  3.  5.100
[... 41 lines cut ...]
The nominal error is that there are no valid streams in the  `crazyswarm_test_video.mp4` file, but this is caused by an error that is hidden in the subprocess that's supposed to generate that video:

```
Traceback (most recent call last):
  File "/home/realm/crazyflie/crazyswarm/ros_ws/src/crazyswarm/scripts/test_videoOutput.py", line 76, in <module>
    videoWriterProcess(sys.argv[1])
  File "/home/realm/crazyflie/crazyswarm/ros_ws/src/crazyswarm/scripts/test_videoOutput.py", line 52, in videoWriterProcess
    timeHelper.sleep(TOTAL_TIME / 2)
  File "/home/realm/crazyflie/crazyswarm/ros_ws/src/crazyswarm/scripts/pycrazyswarm/crazyflieSim.py", line 69, in sleep
    self.visualizer.update(self.t, self.crazyflies)
  File "/home/realm/crazyflie/crazyswarm/ros_ws/src/crazyswarm/scripts/pycrazyswarm/visualizer/visVispy.py", line 100, in update
    mesh.light_dir = (0.1, 0.1, 1.0)
  File "/home/realm/.local/lib/python3.8/site-packages/vispy/util/frozen.py", line 14, in __setattr__
    raise AttributeError('%r is not an attribute of class %s. Call '
AttributeError: 'light_dir' is not an attribute of class <Mesh at 0x7faa8f0082b0>. Call "unfreeze()" to allow addition of new attributes
```

## Additional info

Commenting out lines 100-102 in `visVispy.py` resolves the issue (all three lines seem to be causing issues). There is probably a better fix than this.
```
mesh.light_dir = (0.1, 0.1, 1.0)
mesh.shininess = 0.01
mesh.ambient_light_color = [0.5] * 3
```

jpreiss, 2022-03-09:
Somehow you must have a different VisPy issue than we use for testing/CI, so we will not be able to reproduce this.

Instead of commenting out those lines, can you add `mesh.unfreeze()` before executing them (following the error message suggestion) and try again?

jpreiss, 2022-03-09:
And if it works, please submit a PR :)

dawsonc, 2022-03-11:
Hi! I am the same person as @realm-robot-ws (that's just the account we use on the workstation in our lab). Just chiming in so I get updates on this thread.

closed 2022-03-11 by jpreiss