USC-ACTLab/crazyswarm #638: Enable collision avoidence

Mardank, 2022-04-29:
Hey all,

currently, I'm trying to set up crazyswarm on my system. We are using lighthouse V1 setup with the crazyflie(2.1). I have been able to set up and successfully fly with more than two drones. Now I want to test the onboard collision avoidance with two or more drones. Down below there is a little demo to test the onboard collision avoidance feature. Unfortunately, the collision avoidance didn't work in the experiments. However, in the simulation it works as intended.

Is there anything special I have to be aware of when using it?

Kind regards,
Markus

    #!/usr/bin/env python
    
    import numpy as np
    from pycrazyswarm.crazyswarm_py import Crazyswarm 

    Z = 0.5  # Takeoff altitude.
    
    def main():
        swarm = Crazyswarm()
        timeHelper = swarm.timeHelper
        allcfs = swarm.allcfs
        cfs =  allcfs.crazyflies
    
        swarm.allcfs
        cf_one = cfs[0]
        cf_three = cfs[1]
    
        cf_one.enableCollisionAvoidance([cf_three], np.array([0.3, 0.3, 0.4]))
        cf_three.enableCollisionAvoidance([cf_one], np.array([0.3, 0.3, 0.4]))
    
        allcfs.takeoff(targetHeight=Z, duration=2.0)
    
        timeHelper.sleep(4)
        
        cf_one.goTo(cf_one.position(), 0, 6)
        cf_three.goTo(cf_one.position(), 0, 6)
    
        timeHelper.sleep(7)
    
        allcfs.land(targetHeight=0.06, duration=2.0)
    
        timeHelper.sleep(3)
    
    
    if __name__ == "__main__":
        main()

sAz-G, 2022-05-13:
Hi,

I am having the same problem. I am also using the V1 system and the same Crazyflies. However, I just tried to run the provided script [collisionAvoidance.py](https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/scripts/collisionAvoidance.py) and it did not work. The drones fly, but it does not look like they are avoiding each other.

I followed the installation instructions on [this website](https://crazyswarm.readthedocs.io/en/latest/index.html) and configured the system accordingly.

jpreiss, 2022-05-16:
for collision avoidance to occur, the CFs must be on the same radio channel.