USC-ACTLab/crazyswarm #660: RViz not displaying TF frames for all active Crazyflies

coenwerem, 2022-06-27:
Hi, 

Using the Lighthouse Positioning System, with the `enable_logging` and `enable_logging_pose` parameters in the `hover_swarm.launch` file set to True, I am only able to track the poses of 2 out of 4 (chooser.py, allCrazyflies.yaml) active Crazyflies (see the tfframes_axes.png file). On RViz, I can also see only 3 CF frames checked under the TF display (see the tfframes_rviz.png file). Running `rosrun tf view_frames` outputs the following error:

```
Listening to /tf for 5.0 seconds
Done Listening
b'dot - graphviz version 2.43.0 (0)\n'
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/tf/view_frames", line 119, in <module>
    generate(dot_graph)
  File "/opt/ros/noetic/lib/tf/view_frames", line 89, in generate
    m = r.search(vstr)
TypeError: cannot use a string pattern on a bytes-like object
```

I have checked that the `/cf<id>/pose/` topics are active for all 4 CFs, so I am not sure where the problem is coming from. I am trying to implement a leader-follower flight routine, but this issue has prevented me from retrieving the pose of all the CFs using the Crazyswarm `position()` method, causing my script to fail.

Any help will be greatly appreciated. 

Thanks.

| tfframes_axes.png | tfframes_rviz.png |
| --- | ----------- |
| [image]| [image] |

coenwerem, 2022-06-27:
Here is the tf tree:

[image]

Apparently, `tf` is [deprecated](https://github.com/ros/geometry/pull/193), so I had to use `tf2` instead. As you can see from the pdf file, only 2 CF frames are visible.

Thank you.

whoenig, 2022-06-27:
We update tf whenever we publish a `/cf<id>/pose` message, see https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/src/crazyswarm_server.cpp#L708-L714. 

You could directly subscribe in your script to `/cf<id>/pose` (rather than using the position() method), if these topics give you all the information that you need. My guess is that some of the pose topics might not be updated at a reasonably high rate due to radio bandwidth limitations. If that's the case (check with `ros topic hz`), then switching to two radios (and two channels) might improve the situation.

coenwerem, 2022-06-27:
Thanks for the response, @whoenig. I tried directly subscribing to the `cf<id>/pose` topics, but kept getting zero values, and couldn't get my subscriber to work (a problem on my part and not the package because `rostopic echo cf<id>/pose` works). As a result, I had to resort to using the position() method to expedite things. 

You are right. Using `rostopic hz /cf<id>/pose` returns `no new messages` for `/cf1/pose/` and `/cf2/pose` and actual values for `cf3` and `cf4` pose topics. I will try switching to two radios with individual channels to see if I get better results. Thanks!

coenwerem, 2022-06-27:
Using two Crazyradios with different channels fixed the tf frames issue. Thank you, @whoenig!

[image]

closed 2022-06-27 by coenwerem