USC-ACTLab/crazyswarm #690: hover_swarm.launch crash

fbenti, 2022-09-25:
I've followed the installation guide, however I'm not able to successfully launch the hover_swarm.launch. Here below, my setup:

hover_launch.swarm

<?xml version="1.0"?>

<launch>

  

  <rosparam command="load" file="$(find crazyswarm)/launch/crazyflieTypes.yaml" />

  <rosparam command="load" file="$(find crazyswarm)/launch/crazyflies.yaml" />

  <node pkg="crazyswarm" type="crazyswarm_server" name="crazyswarm_server" output="screen">

    <rosparam>

      # Logging configuration (Use enable_logging to actually enable logging)

      genericLogTopics: ["log1"]

      genericLogTopicFrequencies: [10]

      genericLogTopic_log1_Variables: ["stateEstimate.x", "ctrltarget.x"]

      # firmware parameters for all drones (use crazyflieTypes.yaml to set per type, or

      # allCrazyflies.yaml to set per drone)

      firmwareParams:

        commander:

          enHighLevel: 1

        stabilizer:

          estimator: 2 # 1: complementary, 2: kalman

          controller: 2 # 1: PID, 2: mellinger

        ring:

          effect: 16 # 6: double spinner, 7: solid color, 16: packetRate

          solidBlue: 255 # if set to solid color

          solidGreen: 0 # if set to solid color

          solidRed: 0 # if set to solid color

          headlightEnable: 0

        locSrv:

          extPosStdDev: 1e-3

          extQuatStdDev: 0.5e-1

        kalman:

          resetEstimation: 1

      # tracking

      motion_capture_type: "none" # one of none,vicon,optitrack,qualisys,vrpn

      object_tracking_type: "motionCapture" # one of motionCapture,libobjecttracker

[... 392 lines cut ...]
NODES
  /
    crazyswarm_server (crazyswarm/crazyswarm_server)

auto-starting new master

process[master]: started with pid [6179]

ROS_MASTER_URI=http://localhost:11311

setting /run_id to 20c5f3d8-3ca9-11ed-bc07-65cb421aff1f

process[rosout-1]: started with pid [6193]

started core service [/rosout]

process[crazyswarm_server-2]: started with pid [6200]

[crazyswarm_server-2] process has died [pid 6200, exit code -11, cmd 

/home/filippo/Documents/crazyflie/crazyswarm/ros_ws/devel/lib/crazyswarm/crazyswarm_server __name:=crazyswarm_server __log:=/home/filippo/.ros/log/20c5f3d8-3ca9-11ed-bc07-65cb421aff1f/crazyswarm_server-2.log].

log file: /home/filippo/.ros/log/20c5f3d8-3ca9-11ed-bc07-65cb421aff1f/crazyswarm_server-2*.log

```

fbenti, 2022-09-25:
This is output error when I launch it in debug mode.

`RLException: Roslaunch got a 'No such file or directory' error while attempting to run:

xterm -e gdb --args /home/filippo/Documents/crazyflie/crazyswarm/ros_ws/devel/lib/crazyswarm/crazyswarm_server __name:=crazyswarm_server __log:=/home/filippo/.ros/log/1ae9200a-3cab-11ed-bc07-65cb421aff1f/crazyswarm_server-2.log

Please make sure that all the executables in this command exist and have
executable permission. This is often caused by a bad launch-prefix.
The traceback for the exception was written to the log file
`

closed 2022-10-11 by fbenti

powehi2020, 2022-10-20:
hi~ do you resolve it?