USC-ACTLab/crazyswarm #699: Cannot get CF position when flying more than 3 CF

Dhipz, 2022-10-14:
### Discussed in https://github.com/USC-ACTLab/crazyswarm/discussions/698

<div type='discussions-op-text'>

<sup>Originally posted by **Dhipz** October 13, 2022</sup>
Hi, so i'm trying to get the CF position (x, y, z, coordinates) while flying and i was able to do it for up to 3 CF flying simultaneously. But every time i tried using more than 3 CF only 3 of the CF showed up on the rviz and the code i'm running always gives this errors.

```
/home/user/crazyswarm/ros_ws/src/crazyswarm/scripts/pycrazyswarm/crazyflie.py:590: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  cfg = yaml.load(crazyflies_yaml)
Warning: No joystick found!
Traceback (most recent call last):
  File "code.py", line 16, in <module>
    c_p = cf.position()
  File "/home/user/crazyswarm/ros_ws/src/crazyswarm/scripts/pycrazyswarm/crazyflie.py", line 337, in position
    self.tf.waitForTransform("/world", "/cf" + str(self.id), rospy.Time(0), rospy.Duration(10))
  File "/opt/ros/noetic/lib/python3/dist-packages/tf/listener.py", line 76, in waitForTransform
    raise tf2_ros.TransformException(error_msg or "no such transformation: \"{}\" -> \"{}\"".format(source_frame, target_frame))
tf2.TransformException: canTransform: source_frame cf3 does not exist.
```

This is the test code i used
```py
#!/usr/bin/env python

import numpy as np

from pycrazyswarm import *

swarm = Crazyswarm()
timeHelper = swarm.timeHelper
allcfs = swarm.allcfs

for cf in allcfs.crazyflies:
    cf.takeoff(targetHeight=1.0, duration=2.0)
timeHelper.sleep(3.0)
for cf in allcfs.crazyflies:
    c_p = cf.position()
    print("ID = ", cf.id, " | POS = ", c_p)
    curr_pos = [c_p[0]-0.5,c_p[1],c_p[2]]
    cf.goTo(curr_pos,0,2)
timeHelper.sleep(2.5)
allcfs.land(targetHeight=0.06, duration=2.0)
timeHelper.sleep(3.0)
```

My setup used 2 Lighthouse V2, default CF with lighthouse deck, and all the launch settings are default with the logging enabled and `motion_capture_type` set to none</div>

knmcguire, 2022-11-28:
Hi! 

I can't help you with this particular problem, but if you are not married with ROS1, you should look at the ROS2 port: https://github.com/IMRCLab/crazyswarm2. The cflib backend should have a more stable retrieval of positions with LH v2? Also there is more support over there as well.

Dhipz, 2022-11-28:
Thank you for the reply. Well i somewhat solved it by using 3 channels and Crazyradios for each 2 CF, not the best solution but that will do. 
I have tried Crazyswarm 2 a bit but it seems that you need to somewhat  fully understand ROS to use it? And the guide page is not as complete as Crazyflie 1 (e.g. where do i put my python script?). I do hope that there was a notification of some sort on Crazyswarm 1 guide page and on this repo that there is Crazyswarm 2 in development so i could learn it at the earlier stage of my project.

Dhipz, 2022-11-28:
Thank you for the reply. Well i somewhat solved it by using 3 channels and Crazyradios for each 2 CF, not the best solution but that will do.
I have tried Crazyswarm 2 a bit but it seems that you need to somewhat fully understand ROS to use it? And the guide page is not as complete as Crazyflie 1 (e.g. where do i put my python script?). I do hope that there was a notice of some sort on Crazyswarm 1 guide page and on this repo that there is Crazyswarm 2 in development so i could learn it at the earlier stage of my project. Nonetheless thank you so much for all the work you, bitcraze team, and crazyswarm team have done!

Sorry about opening and closing the issue, i miss click