USC-ACTLab/crazyswarm #733: Drone state estimation drifting significantly for static drone

nl22-nmsu, 2023-03-31:
Hi, I've been using crazyswarm for a couple months now and just recently I've had the issue where the drone's position drifts quite a lot. My best guess is that it is maybe an issue in the state estimation but I'm not sure how to fix it. The vicon position is stable when I use "roslaunch crazyswarm mocap_helper.launch" to look at it. When I echo the topic "/cf1/pose" I can see the drift in the x and y coordinates. 

Any help on solving this would be awesome.

whoenig, 2023-03-31:
Sounds like you might use a different radio that doesn't have the custom firmware? This drifting is an indication that broadcast communication doesn't work, which requires the custom radio firmware.

nl22-nmsu, 2023-03-31:
It looks like it was the radio firmware. The large positional drift is fixed, however while running the hello_world script the crazyflie will start off fine but then starts to spiral outward and eventually crashes.

nl22-nmsu, 2023-03-31:
To add to my last comment. I printed the rate that the /cf1/pose topic is being updated and it's only about 2hz.

nl22-nmsu, 2023-04-17:
Is there a solution to the slow CF position updates? Has this problem been encountered before?

nl22-nmsu, 2023-04-19:
I solved this by reinstalling crazyswarm.

closed 2023-04-19 by nl22-nmsu