USC-ACTLab/crazyswarm #747: issue with testing hello_world.py and hover_swarm.launch

jihoonsuh9, 2023-07-04:
Hi, I am following this tutorial (https://crazyswarm.readthedocs.io/en/latest/tutorials/tutorials.html).

I am using a single crazyflie, and seems to me that configurations are correct with the proper channel and ID (checked in cfclient), and specified them in the .yaml files.

When I run 

roslaunch crazyswarm hover_swarm.launch

I get the following error:

... logging to /home/js65935/.ros/log/01405092-1aa9-11ee-97c2-4123cfa398a9/roslaunch-ase-a38131-lt-48081.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ase-a38131-lt:42077/

SUMMARY
========

PARAMETERS
 * /crazyflieTypes/CF21SingleMarker/batteryVoltageWarning: 3.8
 * /crazyflieTypes/CF21SingleMarker/batteryVoltateCritical: 3.7
 * /crazyflieTypes/CF21SingleMarker/bigQuad: False
 * /crazyflieTypes/CF21SingleMarker/dynamicsConfiguration: 0
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/i_range_z: 0.4
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kR_xy: 70000
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kR_z: 60000
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kd_omega_rp: 200
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kd_xy: 0.2
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kd_z: 0.4
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/ki_xy: 0.05
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/ki_z: 0.05
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kp_xy: 0.4
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kp_z: 1.25
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kw_xy: 20000
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kw_z: 12000
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/mass: 0.032
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/massThrust: 132000
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/kalman/mNGyro_rollpitch: 0.1
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/kalman/mNGyro_yaw: 0.1
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/kalman/pNAcc_xy: 1.0
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/kalman/pNAcc_z: 2.0
 * /crazyflieTypes/CF21SingleMarker/markerConfiguration: 3
 * /crazyflieTypes/default/batteryVoltageWarning: 3.8
 * /crazyflieTypes/default/batteryVoltateCritical: 3.7
 * /crazyflieTypes/default/bigQuad: False
 * /crazyflieTypes/default/dynamicsConfiguration: 0
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_z: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_xy: 70000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_z: 60000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_omega_rp: 200
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_xy: 0.2
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_z: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_xy: 0.05
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_z: 0.05
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kp_xy: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kp_z: 1.25
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kw_xy: 20000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kw_z: 12000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/mass: 0.032
 * /crazyflieTypes/default/firmwareParams/ctrlMel/massThrust: 132000
 * /crazyflieTypes/default/markerConfiguration: 0
 * /crazyflieTypes/defaultSingleMarker/batteryVoltageWarning: 3.8
[... 161 lines cut ...]
[ INFO] [1688502367.343248814]: [cf1] ESTKALMAN: State out of bounds, resetting
[ INFO] [1688502367.345615248]: [cf1] ESTKALMAN: State out of bounds, resetting
[ INFO] [1688502367.347948814]: [cf1] ESTKALMAN: State out of bounds, resetting
[ INFO] [1688502367.350271225]: [cf1] ESTKALMAN: State out of bounds, resetting
[ INFO] [1688502367.351078131]: [cf1] Requesting parameters...
[ INFO] [1688502367.358564629]: Found variables in cache.
[ INFO] [1688502367.724367144]: [cf1] reqParamTOC: 0.385746 s (279 params)
[ INFO] [1688502367.724385768]: [cf1] Requesting logging variables...
[ INFO] [1688502367.726690758]: Found variables in cache.
[ INFO] [1688502367.726987863]: [cf1] reqLogTOC: 0.002634 s
terminate called after throwing an instance of 'std::runtime_error'
  what():  Could not find ctrltarget.x in log toc!
[crazyswarm_server-2] process has died [pid 48128, exit code -6, cmd /home/js65935/crazyswarm/ros_ws/devel/lib/crazyswarm/crazyswarm_server __name:=crazyswarm_server __log:=/home/js65935/.ros/log/01405092-1aa9-11ee-97c2-4123cfa398a9/crazyswarm_server-2.log].
log file: /home/js65935/.ros/log/01405092-1aa9-11ee-97c2-4123cfa398a9/crazyswarm_server-2*.log

I could get around this issue by changing a line within hover_swarm.launch, specifically,

      enable_logging: True    -->   enable_logging: False

I am not sure if this is critical for the hello_world.py to work properly but this was the only way I found to not get the error above.
Are there other proper ways to get around this issue?

Furthermore, with this temporary fix, when I run hello_world.py, I am not observing the expected behavior. My crazyflie starts up but does not lift up vertically (and eventually crashes to the ground because it tries to fly horizontally).

Could I please get any help with this?

knmcguire, 2023-07-05:
Hi!

I can help out with at least the logging part. Your script is giving an error because it is sayint that your crazyflie does not have the following log variable in the firmware: [ctrltarget.x](https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/api/logs/#ctrltargetx). Have you updated the Crazyflie firmware as the instructions indicate?  https://crazyswarm.readthedocs.io/en/latest/configuration.html

The server needs to stay alive in order to run any other scripts so yes it is important.

And yes you need a positioning of some kind, Your Crazyflie will either not take off properly or crash to the ceiling. It is PID with a feedback loop so that is really necessary.

I believe that there is an emergency service in ROS that you can call upon?

Crazyswarm1 is no longer being maintained regurally so I do advise you to switch to Crazyswarm2 if you can. https://imrclab.github.io/crazyswarm2/

knmcguire, 2023-07-06:
Hmm, it's not part of the logging variables that the  launch file has initialized is it? Or else it is probably initialized as part of the python library api I recon

jihoonsuh9, 2023-07-07:
Yes, it’s the line within the hover_launch file. I was directed to set it
as true but gave me errors, so I just set it to false and the server then
runs.

knmcguire, 2023-07-26:
So, I won't be able to test this out myself, but if you are still having issues with this... I would perhaps suggest to try out Crazyswarm2 instead: https://github.com/IMRCLab/crazyswarm2. Crazyswarm(1) support is no longer updated and there will be more help available for Crazyswarm2 (which is based on ROS2)