USC-ACTLab/crazyswarm #749: Uploading more than one trajectory

Ciiciicii, 2023-07-13:
Hi, I am trying to upload two trajectories to the Crazyflie, where the first trajectory moves diagonally upwards and the second trajectory moves the Crazyflie backwards at constant altitude. The code that I used to upload the trajectories are as follows:

    file_path = os.path.dirname(os.path.realpath(__file__))

    traj_name = f"trajectory_id{id}_m{mode}_s{speed_id}.csv"
    full_traj_path = os.path.join(file_path, f"test_trajectories/{traj_name}")

    traj = uav_trajectory.Trajectory()
    traj.loadcsv(full_traj_path)
    cf.uploadTrajectory(0, 0, traj)
    duration = traj.duration * TIMESCALE

    traj_name2 = f"trajectory_id{id}_m{mode}_s{speed_id}_p2.csv"
    full_traj_path2 = os.path.join(file_path, f"test_trajectories/{traj_name2}")

    traj2 = uav_trajectory.Trajectory()
    traj2.loadcsv(full_traj_path2)
    cf.uploadTrajectory(1, 0, traj2)
    duration2 = traj2.duration * TIMESCALE

The code for executing the trajectories and landing are as follows:

    # Start 1st trajectory of crazyflie
    cf.startTrajectory(0, timescale=TIMESCALE)
    timeHelper.sleep(duration+1.0)

    # Start 2nd trajectory of crazyflie
    cf.startTrajectory(1, timescale=TIMESCALE)
    timeHelper.sleep(duration2+1.0)

    cf.startTrajectory(1, timescale=TIMESCALE, reverse=True)
    timeHelper.sleep(duration2+1.0)

    cf.startTrajectory(0, timescale=TIMESCALE, reverse=True)
    timeHelper.sleep(duration+1.0)
    current_pos = cf.position()
    current_z = current_pos[2]

    land_duration = (current_z - 0.06)/speed
    cf.land(targetHeight=0.06, duration=land_duration)
    timeHelper.sleep(land_duration + 1.0)

The code works as intended in the simulation. However, in actual testing, the Crazyflie moves backwards after hovering which puts it out of bounds of the LPS system. Are there any issues with my code? Thank you so much!

Intended path:
[image]

Actual path:
[image]

whoenig, 2023-07-13:
When you upload the second trajectory, the offset needs to be adjusted to the length of the previous trajectories (think of it like an array, where you are responsible to decide at which offset you are storing something).

```
cf.uploadTrajectory(1, traj1.n_pieces(), traj2)
cf.uploadTrajectory(2, traj1.n_pieces() + traj2.n_pieces(), traj3)
...
```

Ciiciicii, 2023-07-13:
Thank you so much for the response. The Crazyflie now follows the two trajectories. I would suggest to add this information to the Crazyswarm documentation as a guide for other users.

closed 2023-07-13 by Ciiciicii