USC-ACTLab/crazyswarm #751: Crazyflie flips over and crashes immediately

CooperDrones, 2023-07-19:
We have a vicon system  with crazyflie 2.1 and crazyradio pa. We are using kalman and mellinger. But when we attempt to hover with one Crazyflie with a unique marker configuration it crashes almost immediately. Identical to the video in issue #150 . Yet the solutions posed did not work for our issue. rviz and vicon seem work fine with tracking. Latency is only 5ms and connection is fine yet the Crazyflie seems to randomly flip over and crash when any script is run. But the drones works fine when used with only cfclient or via a controller. What are your thoughts?/Ways to debug this? @whoenig 
[image]
rqt_graph when connected to crazyflie and rviz running

whoenig, 2023-07-19:
This is almost always a problem with tracking. Make sure you verify the data you see in the `/cf5/poses` topic (enabling logging_pose in the launch file), which is the state estimate after data fusion. If that diverges, you likely have not upgraded your Crazyradio firmware, yet.

CooperDrones, 2023-07-20:
I logged /cf5/pose using rosbag and looked at the plot using rqt_bag and the real position of the drone seems correct when compared to what was logged on that topic.
[image]

However there seems to be very little data points that were logged(note the large gaps btwn the points). Additionally in the rviz terminals i keep getting these warnings.
[ WARN] [1689867380.730327025]: [cf5] Link Quality low (0.550000)
[ WARN] [1689867381.902608865]: [cf5] Link Quality low (0.600000)
[ WARN] [1689867383.068433885]: [cf5] Link Quality low (0.540000)
[ WARN] [1689867384.239172113]: [cf5] Link Quality low (0.540000)
[ WARN] [1689867385.398271360]: [cf5] Link Quality low (0.600000)
[ WARN] [1689867386.561691594]: [cf5] Link Quality low (0.660000)
And the little led on the drone that shows connection seems to be orange instead of birght green/yellow
Could this mean it could be a connection issue. Maybe dropping packets?
What are your thoughts?

whoenig, 2023-07-21:
Yes, that indicates a suboptimal connection. Most of the time this is related to other devices, such as WiFi or Bluetooth-enabled ones, use the same channel. You can try changing channels to improve the situation.

This is, however, not the cause of your crashes, most likely. Did you verify the orientation in a similar fashion (looking at the orientation, while you manually roll/pitch/yaw the drone around)?

CooperDrones, 2023-07-24:
I tilted the drone on its edge and compared it to its pose shown on rviz and on the vicon system, it shows the same orientation. Is there a better way to carry out this test?

CooperDrones, 2023-07-31:
any thoughts?

whoenig, 2023-07-31:
Still seems like a coordinate system issue. You can try single marker tracking to rule out wrong orientation. If you keep the multi-marker tracking, making sure that positive x is in front (where the antenna is).

CooperDrones, 2023-08-02:
We used a single marker and it flies but after it gets to maybe a foot high it starts to drift around a lot in the air back and forth and crash or flip immediately

CooperDrones, 2023-08-02:
UPDATE: WE FIXED IT!!, turns out the x axis on the drone was not properly aligned with the world. So once we turned it 90 deg and ran it. It worked perfectly. Thank you for the help!

CooperDrones, 2023-08-02:
However it still seems very unstable when hovering and often crashes. thoughts?